Zanduino/BME680 #5: Tempreadings to high

chmasi, 2019-09-09:
Hi
after dealing a while with the BME680 I had the feeling that the Temprature out of the readings seems to be to high. After doing some research I found this from Bosch

> Looking at a reference thermometer, we can see that our temperature is indeed a few degrees to high. Does that mean the temperature sensor inside the BME680 is inaccurate?
> Actually no, it very accurately measures the temperature exactly where it is located on the board. But there also is the issue: our board as most devices contains some heat sources (e.g., MCU, WiFi chip, display, …). This means the temperature on our board is actually higher than the ambient temperature. Since the absolute amount of water in the air is approximately constant, this also causes the relative humidity to be lower on our board than elsewhere in the room.
> As BSEC cannot know in which kind of device the sensor is integrated, we have provide some information to the algorithm to enable it to compensate this offset. In the simplest case, we have to deal with an embedded device with a constant workload and approximately constant self-heating. In such a case, we can simply supply a temperature offset to BSEC which will be subtracted from the temperature and will be used to correct the humidity reading as well.
> 
> from the Integration Guide - Bosch Software Environmental Cluster (BSEC)

Do you have a parameter that this mentioned compensation can be added to the initialisation?
If not is there a plan to integrate it?

Thanks
Martin

SV-Zanshin, 2019-09-09:
I don't have anything like that in the library yet. The idea of adding a compensation value to the class initialization is a possibility, as is a separate call which might also allow for a "calibration" to be done if the actual temperature outside the enclosure is known; assuming that the delta between inside and outside remains identical across the expected temperature range.

SV-Zanshin, 2020-05-09:
I'd forgotten this issue until now, when I'm reviewing the library. I will look into how to implement an "offset" to the raw temperature readings - this will be done either at initialization or through an additional function, i.e. "adjustTemperature()" which will apply an adjustment value to all temperature readings.

SV-Zanshin, 2020-05-10:
I initially added 4 functions, "offsetTemperature()", "offsetHumidity()", "offsetPressure()" and "offsetGas()" to the library and then changed my mind. I decided that any adjustment of the raw values is application-dependent and not really a matter for the library to handle. So the library will continue to deliver raw measurements and the user program should adjust those values accordingly instead of making the library more complicated than it needs to be by handling it there.

Hence I'll close this issue - but I remain open to suggestions if you feel that this should be implemented in the library after all.

chmasi, 2020-05-10:
If you look at the port from Bosch itself (
https://github.com/BoschSensortec/BSEC-Arduino-library  ), even the
original library from Bosch has a functionality to set a temperature
offset, this setting is also affecting the calculation of humidity, so no
need for the others in my opinion.
Maybe you think again, and hopefully change your mind again.

Freundliche Grüsse
Martin Sickel
______________________________________________

Oberdorfstr. 28
78337 Öhningen
+49 7735 3551
martin@die-sickels.de

SV-Zanshin, 2020-05-10:
I am looking at the documentation again from Bosch for the BME680 (I see that they have a newer version than what I originally used). The computations in the library for the temperature actually return the compensated values rather than the raw values. Compare the formula used at lines 345-350 in the Zanshin_BME680.cpp library with that on page 17 of the https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bme680-ds001.pdf datasheet.

SV-Zanshin, 2020-05-10:
The BST-BME680-Integration-Guide (file:///C:/Users/Arnd/Downloads/BST-BME680-Integration-Guide-AN008-47.pdf) references an offset in the initialization code for "bsec_iot_init", which is unfortunately not implemented in their Arduino version and the corresponding "begin()" method has no such parameter for compensation. I will have to see if I can find the source code in the BSEC software where they apply that offset and how they change the relative humidity reading; if I can find that code then it could be added to the library - but at the moment I am uncertain of how it would be applied.

SV-Zanshin, 2020-05-10:
OK, I've downloaded the main BSEC software and am looking into it. The compensation formula is not directly visible to me.

SV-Zanshin, 2020-05-16:
It looks to me as if the compensation already done by the library is what Bosch is referring to, and that the only difference is that a fixed temperature offset can be set in BSEC.

closed 2020-05-17 by SV-Zanshin