Zanduino/BME680 #7: SparkFun Fork?

AndyEngland521, 2020-05-04:
Hi there @SV-Zanshin 

I'd like to start by saying this isn't an issue, more of a request. SparkFun is currently making a breakout for the BME680 and would like to point to your library in our hookup guide. This can sometimes lead to an influx of questions/issues/pull requests on the repository. We are currently considering two options.

1. Fork the library so we are able to update the library on our own.
2. Leave the library as is, updating the library through pull requests.

Although your library is licensed as GNU, we wanted to ask if you had a preference before we move forward.

Thanks in advance, and excellent work,
Andy England

SV-Zanshin, 2020-05-04:
Hello Andy,

 

I am not sure how active the library would be should SparkFun make a breakout and reference it; I do attempt to keep up on requests for the other libraries I’ve put out there. But your mail will have me looking into my closet to make sure I have a BME680 here (I’m currently in lockdown in Germany, and half of my Arduino hardware is elsewhere and out of reach) just in case request do come in.

 

I haven’t looked into the collaboration options on github – perhaps keeping the library as is but allowing others access to fulfil pull requests and other changes. If that doesn’t work then of course the GNU license means there are no restrictions for re-using the code.

 

On a related topic, I’ve been playing around with a AtTiny85 controller for the 3-LED colored rotary encoder which supports I2C, Serial and 1-Wire protocols and does a very smooth 10-bit PWM on the 3 LEDs. I have it working and even had some prototype boards made in China and have been wondering about whether there might be a market for a breakout board for this application.

 

Regards,

 

-Arnd.

AndyEngland521, 2020-05-04:
Collaboration options would be ideal for us as this would allow us to maintain your library if necessary.

As for the encoder, well, [there's definitely a market for that application](https://www.sparkfun.com/products/15083)

SV-Zanshin, 2020-05-05:
Regarding the encoder – I should have researched deeper, although I would have still programmed it.

 

What I have is a bit of “smarter” solution – you send a command consisting of “LED,startPWM,endPWM,speed” to the I2C (or Serial, or 1-Wire) bus and the Rotary encoder takes care of the rest. The speed goes from almost immediate to taking 30 seconds to complete, and the Red, Green and Blue speeds and direction are independent and can be combined for some really wild effects. But the PWM fading is what I really did it for, since using an Arduino 8-Bit PWM really sucks due to each increment being very visible but most Atmel chips only support 1 or 2 16-bit PWM in hardware. So I wrote a solution that does 10-bit PWM on 3 pins while still catering to the interrupt requirements of the communications ports.

 

While designed for the minimal attiny85 it does work on all the other Atmel processors after setting the registers appropriately.

 

I was going to put together a product datasheet for sending to Sparkfun but since this would “compete” with your Qwiic Twist I suppose that is a moot point now. But it was fun for playing around and for getting very acquainted with my oscilloscope since writing a 1-Wire slave device proved very challenging as the timings are very tight for the relatively slow Atmel chips.

 

All the best and I’d be thrilled to see the BME680 library see some use. I have a bit of spare time now working from my flat (I’m currently in lockdown in Bavaria doing consulting work for an insurance company remotely – at least I’m close to some woods to go walking in and the beer is excellent and cheap, the local brew is about €4 for 4x500ml bottles 😊), so I think I might look in my box for a BME680 and see if I can clean up the library prior to it getting published any wider.

 

-Arnd.

SV-Zanshin, 2020-05-05:
I just realized that I haven't been replying to e-mails but to this thread and the comments about the "SmartEncoder" that I've working on aren't really germane. Sorry about that.

What is your time horizon for a SparkFun BME680? I rooted around my box of breakout boards and found a BME680 from a company called BlueDot (www.bluedot.space) and will hook that up and see about cleaning up my library code in preparation for an audience of more than one person :) 
But perhaps, down the road, I could get a prototype of your board to ensure compatibility.

AndyEngland521, 2020-05-06:
@SV-Zanshin can you contact me at andrew.england@sparkfun.com to discuss this further? A nice Bavarian beer sounds delicious as well!

closed 2020-05-06 by AndyEngland521