Zanduino/BME680 #26: Where are the Gas calibration constants used?

stefanrueger, 2020-09-21:
I've browsed through the source code and didn't see where the Gas calibration constants are used. This *must* be a bug. I recommend reviewing the code and comparing against the original published library from Bosch Sensortek.

Stefan

stefanrueger, 2020-09-21:
OK, I know where the problem is: lines 481 and 482 of the .cpp file wongly use the *humidity* constants _H1 ... _H3 instead of the _G1 ... _G3 *gas* constants (this function sets the parameters for the gas measurement). The gas readings between different sensors will be inconsistent, and the temperature settings for the gas sensor hot plate quite wrong.

stefanrueger, 2020-09-21:
There is another problem in the section of lines 410 to 414: (int32_t) 1 << 31 yields the smallest 32 bit signed integer; the comparison on line 411is therefore always true. The code looks like it wants to avoid computational 32-bit overflow. As a consequence the code always loses one bit resolution (rather than only in those cases where the numbers would overflow). Not a biggie. It looks like the code was originally written for 32 bit arithmetics, but that is at loggerheads with the variables actually being 64 bit here? 

Again, I recommend comparing against the published reference implementation from Bosch. Perhaps a good idea to go carefully go through the full code?

stefanrueger, 2020-09-21:
OK - I've just checked the reference implementation at https://github.com/BoschSensortec/BME680_driver 

It's 1 << 30 (not 1 << 31); and yes, in that section of the code it's sufficient to use 32 bit integer variables.

stefanrueger, 2020-09-21:
Variable status in readSensors() is computed, but not used.

SV-Zanshin, 2020-09-27:
Corrected the errors that you found in the source code regarding gas computations and pressure overflow calculations. Good catch.

SV-Zanshin, 2020-09-27:
I've added an optional return value for [getSensorData()](https://github.com/SV-Zanshin/BME680/wiki/getSensorData())

closed 2020-09-27 by SV-Zanshin