Zanduino/BME680 #30: how are the readings done?  sleep mode?

Alain2019, 2020-10-04:
Hi

How is de lebrary doing the readings?

After looking at the source code I have the impression that the sensor is constantly running doing measurements.

Is it possible to let it "sleep" and only measure when needed?

Where in the source code should I look?

SV-Zanshin, 2020-10-05:
Each time a call to [getSensorData()](https://github.com/SV-Zanshin/BME680/wiki/getSensorData())  
 is  made a single measurement is triggered. If no calls to this function are me then no measurements are done. So you control how often readings are done; this is also explained on the wiki page for [getSensorData()](https://github.com/SV-Zanshin/BME680/wiki/getSensorData())

closed 2020-10-05 by SV-Zanshin

SV-Zanshin, 2020-10-05:
That is correct, the first returned set of values would be stale, 15 minutes old, and the second set would be the recent values.