Zanduino/BME680 #31: BME680 Humidity calibration according to datasheet?

Alain2019, 2020-10-06:
Hi

I've seen a possible other calibration in you're code (and those from Bosh, versus the BME680 datasheet.

According the datasheet I would expect :

.h
const uint8_t BME680_BIT_H1_DATA_MSK          = 0xF0;  ///< Mask for humidity  (instead of 0x0F)

.cpp

  _H1 = (uint16_t)(((uint16_t)coeff_arr2[BME680_H1_MSB_REG] << BME680_HUM_REG_SHIFT_VAL) |
                   ((uint16_t)(coeff_arr2[BME680_H1_LSB_REG] & BME680_BIT_H1_DATA_MSK)>>4) );
  _H2 = (uint16_t)(((uint16_t)coeff_arr2[BME680_H2_MSB_REG] << BME680_HUM_REG_SHIFT_VAL) |
                   ((uint16_t)(coeff_arr2[BME680_H2_LSB_REG] & BME680_BIT_H1_DATA_MSK)>>4));

SV-Zanshin, 2020-10-07:
Where in the datasheet did you see the bitmask definitions for the humidity? I just see the two 8-bit registers for LSB and MSB defined.

Looking at the original BSEC source code and the code in my library they are identical in function, and I suspect that there is no error in the BSEC code as that has been in production for a while with many users and the humidity computations would be quite inaccurate if the calibration computations were to be incorrect.

The value of the "BME680_HUM_REG_SHIFT_VAL" is 4, which would correctly correspond to the BME680_BIT_H1_DATA_MSK value of 0x0F which would mask the last 4 bits.

Alain2019, 2020-10-07:
https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME680-DS001.pdf

page 20 table 13

Variable nameRegister address (LSB / MSB)
par_h1 0xE2<7:4> / 0xE3 
par_h2 0xE2<7:4> / 0xE1

a 7:4 mask is according to me 0xF0  (and then >>4 to get the correct position)  ir for 8bit only >> 4  is the same.

Alain2019, 2020-10-07:
BTW. The reason that I started investigating is that I can get only values into the low 80's, even in a complete damp room (water cooker making a lot of humidity.)

I also expect higher values in other locations.

SV-Zanshin, 2020-10-07:
Aha, my version of the documentation doesn't have that chart. Let me look into this. Have you tried making the changes you mentioned in your copy of the library, and have you gotten better / more correct readings?

Alain2019, 2020-10-07:
The changes are -in my case- minimal : _H1 goes from 620 to 623.
_H2 is the same, but that codes seems correct, but optimised. 

Chnge in pressure is 58,63 to 58,5.

closed 2020-10-10 by SV-Zanshin