Zanduino/BME680 #38: Gas Resistance Values are Static

roshik142234, 2022-08-05:
## Expected Behavior

The gas resistance value should change smell is detected. 

## Actual Behavior

The value is static and prints out a negative value in the 2000s. It is a library problem because the gas value changes when using the adafruit BME680 library.

## Steps to Reproduce the Problem

_Explain what needs to be done in order to reproduce the problem._
  1.Make a sketch that includes the Zanshin BME680 library or use the IC2DEMO
  2.Print the gas resistance value in serial monitor.
  3.Run code and open serial monitor.

## Specifications

  - Library Version:1.0.10
  - IDE Version:1.8.19
  - Platform:Arduino IDE
  - Subsystem:esp32 with BME680 sensor.

SV-Zanshin, 2022-08-07:
I can't find my BME680 board here (the other 2 are installed in remote devices), otherwise I could check. I do know that I was getting values during development - but I didn't try to make sense of them as the documentation showed that it wasn't all that easy. 
I'll continue looking for my board - I know I must have it somewhere around here.

SV-Zanshin, 2022-09-04:
I found a board, but it is the 3.3v and all of my devices are 5V, so I have to root around for a level-shifter before I can reproduce the issue.

SmartDucksWorks, 2024-03-16:
Get it saved. it's definitely got some issues.