Zanduino/BME680 #40: UNO R4 Qwiic Support for BME680

baqwas, 2024-07-28:
Hello!

I tried to use your BME680 library for the UNO R4 Qwiic interface to a Sparkfun BME680. I failed miserably!

Your library works fine on the standard I2C interface on the UNO R4 with a BME680. I didn't have to change any statement for your example **I2CDemo.ino** sketch.

However, I need your guidance on making the library work with UNO R4 Qwiic interface, please. At [https://docs.arduino.cc/tutorials/uno-r4-wifi/qwiic/](https://docs.arduino.cc/tutorials/uno-r4-wifi/qwiic/), Arduino suggests the following statements in the setup() section:
```
Wire1.begin();
libraryName.begin( Wire1 );
```
In my _naivety_:

- I took your BME680.h file and replaced all instances of **Wire** to **Wire1**
- I renamed BME680 Class with a suffix, __Qwiic_
- I instantiated a sensor variable with the modified Class definition
- I could only insert Wire1.begin() in the sketch because libraryName.begin() could not be found

Verify/Compile of the modified sketch produced the following error message:
```
/tmp/arduino/sketches/9436EA69531D56D990479CE1FAF17E54/sketch/I2CDemox.ino.cpp.o: In function `loop':
/home/reza/Nextcloud/IoT/Arduino/foxcam/unor4/I2CDemox/I2CDemox.ino:146: undefined reference to `BME680_Class::getSensorData(long&, long&, long&, long&, bool)'
/tmp/arduino/sketches/9436EA69531D56D990479CE1FAF17E54/sketch/I2CDemox.ino.cpp.o: In function `_GLOBAL__sub_I_BME680_Qwiic':
/home/reza/Nextcloud/IoT/Arduino/foxcam/unor4/I2CDemox/I2CDemox.ino:78: undefined reference to `BME680_Class::BME680_Class()'
/tmp/arduino/sketches/9436EA69531D56D990479CE1FAF17E54/sketch/I2CDemox.ino.cpp.o: In function `_GLOBAL__sub_D_BME680_Qwiic':
/home/reza/Nextcloud/IoT/Arduino/foxcam/unor4/I2CDemox/I2CDemox.ino:78: undefined reference to `BME680_Class::~BME680_Class()'
/tmp/arduino/sketches/9436EA69531D56D990479CE1FAF17E54/sketch/I2CDemox.ino.cpp.o: In function `setup':
/home/reza/Nextcloud/IoT/Arduino/foxcam/unor4/I2CDemox/I2CDemox.ino:116: undefined reference to `BME680_Class::begin(unsigned long, unsigned char)'
/home/reza/Nextcloud/IoT/Arduino/foxcam/unor4/I2CDemox/I2CDemox.ino:121: undefined reference to `BME680_Class::setOversampling(unsigned char, unsigned char) const'
/home/reza/Nextcloud/IoT/Arduino/foxcam/unor4/I2CDemox/I2CDemox.ino:122: undefined reference to `BME680_Class::setOversampling(unsigned char, unsigned char) const'
/home/reza/Nextcloud/IoT/Arduino/foxcam/unor4/I2CDemox/I2CDemox.ino:123: undefined reference to `BME680_Class::setOversampling(unsigned char, unsigned char) const'
/home/reza/Nextcloud/IoT/Arduino/foxcam/unor4/I2CDemox/I2CDemox.ino:125: undefined reference to `BME680_Class::setIIRFilter(unsigned char) const'
/home/reza/Nextcloud/IoT/Arduino/foxcam/unor4/I2CDemox/I2CDemox.ino:127: undefined reference to `BME680_Class::setGas(unsigned short, unsigned short) const'
collect2: error: ld returned 1 exit status
```

Obviously, my knowledge of C/C++ leaves a lot to be desired. When I inspected the BME680.h file, I did not see any code for some of the public functions, although the declarations were present.

What changes do I need to make your BME680.h file to leverage Wire1() functions in my sketch to read data from a BME680 sensor on the UNO R4 Qwiic interface? Thanks.

Regards.

SV-Zanshin, 2024-07-29:
I didn't even realize that the R4 was available. Since the underlying hardware is very different from the Atmel processors, the library interface needs to be different as well. I've ordered one for myself and will dig out my BME680 and see about getting it to function correctly with this new hardware.

baqwas, 2024-07-29:
Thanks! Appreciate your efforts in keeping the library current. I love the BME680 ever since I _discovered_ its existence a few years ago.

Regards.

P.S.
I too ignored the R4 until a month ago when I needed a _low-cost_ but contemporary solution for multiple analog pins.

closed 2024-07-29 by baqwas

SV-Zanshin, 2024-08-06:
I just wanted to add to this. I've gotten the R4 and it is a great little device.

But I wrote the BME680 library for the Atmel 8-bit architecture specifically to use a small footprint and mainly to avoid using floating-point arithmetic.  The R4 chipset, on the other, has much more memory and has hardware floating-point; so, using the original Bosch library makes much more sense than adapting my library - I don't see any advantage in doing this.

baqwas, 2024-08-06:
Understand your viewpoint, @SV-Zanshin. Appreciate your feedback.  :saluting_face:  Loved your library for my simplistic needs. The Bosch Sensortec library is beyond my job grade. That is why I had turned to your library in the first place. No worries, just an extra incentive to learn something new now. :wink: 

Warmest regards.